Dilbert66/esphome-vistaECP #157: User Code Error: Keys 0-9 Locked

AlexandreUSA, 2025-08-09:
I updated my VistaAlarm to the latest revision as of 8/9/2025 using the external components @main on an ESP032. I have been running VistaAlarm for years now and without any major issues. 

After my update, everything seems to be working fine as expected expect that I am getting the following messages: "CID_3461: Wrong Code Entry on zone 0 , Partition 1" ... or Partition 3... I do not have partitions 2 or 3. These happen randomly and are fine. However, sometimes, I may get into the User Code Error and that locks my physical keypad and panel. Pretty tough as for the example below, my alarm is set and I can not disarmed it. I got the error after entering my code via the home assistant card.

Also, at one point, my alarm system physical keypads were locked for too many wrong code entries. Nothing was happening at the physical keypad. 

`[09:42:59][I][CMD:816]: [2025-08-09 09:42] F2 31 00 00 00 00 00 65 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AD 
[09:43:00][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 66 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AC 
[09:43:00][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 67 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AB 
[09:43:01][D][vista_alarm:1271][cmdQueueTask]: High water stack level:  1364
[09:43:01][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 60 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B2 
[09:43:01][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 61 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B1 
[09:43:02][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 62 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B0 
[09:43:02][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 63 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AF 
[09:43:02][I][CMD:816]: [2025-08-09 09:43] F7 00 00 0F 10 08 00 8C 08 02 00 00 41 52 4D 45 44 20 2A 2A 2A 53 54 41 59 2A 2A 2A 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 86 00 00 00 
[09:43:02][I][vista_alarm:1691]: Partition: 01
[09:43:02][I][vista_alarm:1711]: Prompt: ARMED ***STAY***
[09:43:02][I][vista_alarm:1712]: Prompt:                 
[09:43:02][I][vista_alarm:1713]: Beeps: 0
[09:43:03][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 64 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AE 
[09:43:03][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 65 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AD 
[09:43:04][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 66 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AC 
[09:43:04][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 67 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AB 
[09:43:05][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 60 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B2 
[09:43:05][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 61 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B1 
[09:43:06][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 62 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B0 
[09:43:06][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 63 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AF 
[09:43:07][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 64 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AE 
[09:43:07][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 65 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AD 
[09:43:08][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 66 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AC 
[09:43:08][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 67 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AB 
[09:43:09][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 60 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B2 
[09:43:09][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 61 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B1 
[09:43:10][I][CMD:816]: [2025-08-09 09:43] FB 02 20 81 62 
[09:43:10][I][EXT:816]: [2025-08-09 09:43] 00 24 00 DC 00 00 00 00 00 00 00 00 74 
[09:43:10][I][RAW:816]: [2025-08-09 09:43] 00 24 00 DC 
[09:43:10][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 62 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B0 
[09:43:10][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 63 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AF 
[09:43:11][I][CMD:816]: [2025-08-09 09:43] FB 02 25 82 5C 
[09:43:11][I][EXT:816]: [2025-08-09 09:43] 00 21 03 DC 00 00 00 00 00 00 00 00 74 
[09:43:11][I][RAW:816]: [2025-08-09 09:43] 00 21 03 DC 
[09:43:11][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 64 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AE 
[09:43:11][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 65 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AD 
[09:43:12][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 66 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AC 
[09:43:12][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 67 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AB 
[09:43:12][I][CMD:816]: [2025-08-09 09:43] F7 00 00 0F 10 08 00 8C 08 02 00 00 41 52 4D 45 44 20 2A 2A 2A 53 54 41 59 2A 2A 2A 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 86 00 00 00 
[09:43:12][I][vista_alarm:1691]: Partition: 01
[09:43:12][I][vista_alarm:1711]: Prompt: ARMED ***STAY***
[09:43:12][I][vista_alarm:1712]: Prompt:                 
[09:43:12][I][vista_alarm:1713]: Beeps: 0
[09:43:13][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 60 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B2 
[09:43:13][I][CMD:816]: [2025-08-09 09:43] F2 31 00 00 00 00 00 61 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 B1 
[09:43:13][D][vista_alarm:654]: Writing keys: FIRST DIGIT DELETED to partition 1
[09:43:14][I][CMD:816]: [2025-08-09 09:43] F6 13 
[09:43:14][E][CHK:814]: [2025-08-09 09:43] 78 13 
[09:43:14][I][CMD:816]: [2025-08-09 09:43] F7 00 00 0F 10 08 00 8C 08 02 00 00 41 52 4D 45 44 20 2A 2A 2A 53 54 41 59 2A 2A 2A 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 06 00 00 00 
[09:43:14][I][vista_alarm:1691]: Partition: 01
[09:43:14][I][vista_alarm:1711]: Prompt: ARMED ***STAY***
[09:43:14][I][vista_alarm:1712]: Prompt:                 
[09:43:14][I][vista_alarm:1713]: Beeps: 0
[09:43:14][I][CMD:816]: [2025-08-09 09:43] F9 43 02 53 6F 
[09:43:14][E][CHK:814]: [2025-08-09 09:43] C3 
[... cut ...]
[09:44:01][I][CMD:816]: [2025-08-09 09:44] F2 31 00 00 00 00 00 65 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AD 
[09:44:01][I][CMD:816]: [2025-08-09 09:44] F2 31 00 00 00 00 00 66 6D 0D 45 43 F5 EC 55 73 65 72 20 43 6F 64 65 20 45 72 72 6F 72 3A 20 4B 65 79 73 20 30 2D 39 20 4C 6F 63 6B 65 64 00 31 00 35 AC 
`

What would be the best path to troubleshoot this one?

[image]

Also, most likely unrelated but what is this message in the log for:

[09:42:31][D][vista_alarm:1271][cmdQueueTask]: High water stack level:  1364

AlexandreUSA, 2025-08-09:
2nd log when I tried to disarmed the alarm again. Could not do it. Restarted the panel and used the physical keypad.

[Log 2nd.txt](https://github.com/user-attachments/files/21698430/Log.2nd.txt)

AlexandreUSA, 2025-08-19:
Hi, I tried the "0" in the auiaddr but the trouble continues. The first night, I got two wrong code messages but no locked out at the keypads.  I then left for a business trip and I left the ESP32 disconnected. No issues during my trip. I finally reconnected today and tried to figure out what is going on. The log is attached. Here are my steps:
Power up the ESP32, all good in Homeassistant. I tried to arm the system a couple times via the home assistant virtual keypad, no luck. I then went to the physical keypad in the Foyer and arm the system there. All good. I then went back to my virtual keypad in HA and I tried to disarmed the system. The code was accepted and the system disarmed. My issues started right away after that.  Got a  
[14:59:01][D][text_sensor:069]: 'Lrr Msg': Sending state 'CID_3461: Wrong Code Entry on zone 0 , Partition 2'
I never enable the Partition 2 in my system. or in home assistant, then a 
[14:59:03][D][text_sensor:069]: 'System Status': Sending state 'triggered'
[14:59:03][D][binary_sensor:026]: 'Fire': New state is ON
The physical started to beep and the keys were locked. That freaked me out and I went down the panel to remove power/battery to restart. 
Got also other wrong code entry but now on Partition 1
[14:59:23][D][text_sensor:069]: 'Lrr Msg': Sending state 'CID_3461: Wrong Code Entry on zone 0 , Partition 1'

[VistaAlarm.txt](https://github.com/user-attachments/files/21865917/VistaAlarm.txt)

Dilbert66, 2025-08-19:
Ok, i'll need to see your configuration because the issue your having doesnt make sense. I can't duplicate it.

Dilbert66, 2025-08-19:
It looks like you have a whole pile of keypad programmed in the system
[14:59:15][I][CMD:816]: [2025-08-19 14:59] F6 13 
[14:59:15][I][CMD:816]: [2025-08-19 14:59] F6 14 
[14:59:15][I][CMD:816]: [2025-08-19 14:59] F6 15 
[14:59:15][I][CMD:816]: [2025-08-19 14:59] F6 16 
[14:59:15][I][CMD:816]: [2025-08-19 14:59] F6 17 
[14:59:14][I][CMD:816]: [2025-08-19 14:59] F6 10 
[14:59:09][I][CMD:816]: [2025-08-19 14:59] F6 11 
[14:59:09][I][CMD:816]: [2025-08-19 14:59] F6 02 
[14:59:14][I][CMD:816]: [2025-08-19 14:59] F6 12 
From address 16 to 23 including address 2 which is an aui address.

AlexandreUSA, 2025-08-19:
I have been modifying the vistaalarm configuration file for years, so I may have messed up something in the last update. I did not have issues before that recent change as I was on the latest version of VistaAlarm and ESPhome since the release a few months ago. I have attached my configuration file for your reference. 

16. Foyer Keypad 6160
17.  Laundry Keypad 6160 
18. Master Bedroom Keypad 6160RF
19. VistaAlarm ESP32
07. 4229 Zone 9-16

I do not have touchscreen keypads or TOTAL Connect but I do have a monitoring company. 

Thank you so much for the support. You have been amazing at maintaining this integration and I cannot thank you enough.

[vistaalarm.yaml](https://github.com/user-attachments/files/21868274/vistaalarm.yaml)

Dilbert66, 2025-08-20:
Honestly I really can't see what is causing the errors.  the user code errors would occur if an invalid user code and cmd was sent more than 6 times.  It would then lock the panel for about 15 minutes.  I don't see any invalid data being sent .   One thing I did notice though is that you removed all the F6 related responses to your code entries.  That kind of skews the  log a bit for understanding the sequence.  It would be more helpfull if you just put XX where the code digits are (or your best guess) so I can see if the respsonses are sent as they should be.   You can also send the file directly to me at this email address (dilbert341 at gmail.com)  Change the at to @.  and this way your log won't be in public view.

AlexandreUSA, 2025-08-20:
> One thing I've seen though is that a bad connection to the esp such as a bad ground , etc can cause spurious data on the lines and cause odd behaviour. The fact that the fire sensor was triggered is also odd here. Normally a bad entry would not do that which is why I'm leaning more towards a connection issue. Anyhow, if you can email me another log of an arm /disarm from the virtual keypad, that would be helpful.

Hi,
Let me check for more data. I only replaced the single digit with the XX. Nothing else has been removed. I am I missing some lines? I will check for connection issue but I did not touch the esp32 for a while and the problem came right after I upgraded my config file with the new changes. Could it also be related to the home assistant card? The mystery continues.

Alex

AlexandreUSA, 2025-10-16:
I am still having my issue. New board, new ESP32, new resistors, and new 4N35. Still the same problem with my system. Any ideas on what to look for next? Could it really be hardware when my system was working perfectly fine until the large SW update in August? How can I have the exact same issue with my new config as the whole circuit is new. When I unplug the ESP32, my alarm system works perfectly. When I plug in the ESP32 and remove the connection of the red wire (mostly disabling sending data), everything is as normal too. I can monitor without issue. Looking for help here as I am out of ideas.

New log: 
`

[logs_vistaalarm_logs (7).txt](https://github.com/user-attachments/files/22956721/logs_vistaalarm_logs.7.txt)

`

Dilbert66, 2025-10-16:
I'm still not sure what the issue is exactly that is causing it but I do see some very odd behaviour in the logs.  I  have a suspicion but I need to have you test out this latest version I've just pushed to see if that's the case.

AlexandreUSA, 2025-10-30:
> I've pushed on update on main and dev that might address this issue. From what I see, the panel is getting a distorted header byte from the esp on an f6 response (send cmd). I highly suspect this is due to the esp sending the response too quickly back to the panel before it's ready. This is most likely due to the fact that I moved a lot of timing critical code to it's own cpu task which removed a lot of inherent delays. I added a delay before the response to alleviate this. Post back with results or logs if the issue is still happening.

Thanks. I will try the new code today.

Alex